martin-ger/esp_slip_router #3: Why the 115200 baud limit?

gazsp, 2018-03-03:
I thought the ESP8266 UART could go much faster than this?

http://www.esp8266.com/viewtopic.php?f=6&t=1711

Nice work on this, by the way :-)

martin-ger, 2018-03-03:
There is no built-in limit for the ESP. Just enter:
"set bitrate xxx"
and it will configure the HW UART directly (after "save" and reboot).

I have just realized that my Linux Laptop doesn't manage more than 115200 on the USB-serial converter - this is my practical limit for tests...

gazsp, 2018-03-05:
Ah, OK. Thanks for the info.

I have a USB cable with the same issue. I replaced it with an FTDI cable, which supports up to 1Mbps.

closed 2018-03-05 by gazsp

SergeyPL, 2019-02-25:
The slattach has the 115200 limit. It need to be recompiled from sources after changing the available speed list. I tested in Debian 9 460800 - it works.

martin-ger, 2019-02-25:
That's great - you didn't need to change anything on the esp_slip_router sources, right?

SergeyPL, 2019-02-25:
That is right.