martin-ger/esp_slip_router #4: Trouble compiling in softuart.c: 'ets_isr_t' undeclared - SDK 2.1.0 problem?

ulno, 2018-04-20:
This software is just what I was looking for - congrats and huge thanks already. It might solve tons of problems in my classes.

I would like to use (and probably extend) this firmware to create a reliable accesspoint for my IoT project ulnoiot (also here on github). It's nearly impossible to reliably source (multiple) wifi sticks which support accesspoint mode nowadays (internal wifi of raspberry pi is a positive exception, however, you can't buy the pi zero w in quantities sufficient for classrooms), and I have plenty of esp8266s anyway.

I have no trouble compiling esp_wifi_repeater, but compiling this, I get the following (independent on how I set DEBUG_SOFTUART or not:
```
$ make
CC driver/softuart.c
In file included from driver/softuart.c:1:0:
driver/softuart.c: In function 'Softuart_Init':
driver/softuart.c:158:25: error: 'ets_isr_t' undeclared (first use in this function)
   ETS_GPIO_INTR_ATTACH((ets_isr_t)Softuart_Intr_Handler, s);
                         ^
/home/ulno/ulnoiot/.local/external/firmware/esp-open-sdk/sdk/include/ets_sys.h:79:36: note: in definition of macro 'ETS_GPIO_INTR_ATTACH'
     ets_isr_attach(ETS_GPIO_INUM, (func), (void *)(arg))
                                    ^
driver/softuart.c:158:25: note: each undeclared identifier is reported only once for each function it appears in
   ETS_GPIO_INTR_ATTACH((ets_isr_t)Softuart_Intr_Handler, s);
                         ^
/home/ulno/ulnoiot/.local/external/firmware/esp-open-sdk/sdk/include/ets_sys.h:79:36: note: in definition of macro 'ETS_GPIO_INTR_ATTACH'
     ets_isr_attach(ETS_GPIO_INUM, (func), (void *)(arg))
                                    ^
driver/softuart.c:158:35: error: expected ')' before 'Softuart_Intr_Handler'
   ETS_GPIO_INTR_ATTACH((ets_isr_t)Softuart_Intr_Handler, s);
                                   ^
/home/ulno/ulnoiot/.local/external/firmware/esp-open-sdk/sdk/include/ets_sys.h:79:36: note: in definition of macro 'ETS_GPIO_INTR_ATTACH'
     ets_isr_attach(ETS_GPIO_INUM, (func), (void *)(arg))
                                    ^
Makefile:142: recipe for target 'build/driver/softuart.o' failed
make: *** [build/driver/softuart.o] Error 1
```

My wild guess is that something changed from the esp sdk from 2.0 to 2.1, but maybe it has a different cause. It's probably not a hard thing to fix, but would be happy for some guidance or pointers.

Cheers,
ulno

martin-ger, 2018-04-20:
Strange - for me it compiles (aside from a nonsense "#include <cli.h>" in user_main.c).
ets_isr_t is declared in "ets_sys.h":
`67: typedef void (* ets_isr_t)(void *);`
and this in included in the first line. Could you check that?

ulno, 2018-04-20:
I have to back-paddle a bit - wondering myself i deleted my esp sdk and re-downloaded - and as usual I can't compile the newest version - always happens. So, I come back when I can compile that again.

ulno, 2018-04-20:
Compiles now - my bad. Closing this.

closed 2018-04-20 by ulno