martin-ger/esp_slip_router #5: Can't show console command

hanyuzzw, 2018-06-13:
Excuse me! Follow your steps, whether I configure the SLIP interface with : 

        `IP4_ADDR(&netmask, 255, 255, 255, 0);
         IP4_ADDR(&gw, 127, 0, 0, 1);
         netif_add (&sl_netif, &config.ip_addr, &netmask, &gw, &int_no, slipif_init, ip_input);
         netif_set_up(&sl_netif);`

or with:

        `IP4_ADDR(&netmask, 255, 255, 255, 0);
         IP4_ADDR(&gw, 192, 168, 240, 2);
         netif_add (&sl_netif, &config.ip_addr, &netmask, &gw, &int_no, slipif_init, ip_input);
         netif_set_up(&sl_netif);`

it always can't show the console CMD after I run:

       `sudo slattach -p slip -s 115200 /dev/ttyUSB0 & sudo ifconfig sl0 192.168.240.2 pointopoint 
        192.168.240.1 up mtu 1500
        sudo telnet 192.168.240.1 7777`

why?

martin-ger, 2018-06-13:
But you also set
`IP4_ADDR(&ipaddr, 192, 168, 240, 1);`
on the ESP?

The telnet only works if you have a server running on the ESP. If you are referencing the simple example, the only thing you can do is pinging. Does the user_simple work for you?

hanyuzzw, 2018-06-13:
yes, the `config.ip_addr` is 192.168.240.1, and I compiled with the user not the user_simple.  I succeed one time, but I can't reappear it any more .   
And if I use user_simple, the esp led is blinking all the time, and
    `$ ping 192.168.240.1
     PING 192.168.240.1 (192.168.240.1) 56(84) bytes of data.
     ping: sendmsg: No buffer space available
     ping: sendmsg: No buffer space available
     ping: sendmsg: No buffer space available
     ping: sendmsg: No buffer space available
     ping: sendmsg: No buffer space available
     ping: sendmsg: No buffer space available`

but ping other ip is ok.

the configuration of slip interfance is:
  
       `sl0       Link encap:Serial Line IP  
                   inet addr:192.168.240.2  pointtopoint :192.168.240.1  netmask:255.255.255.255`

martin-ger, 2018-06-14:
Just uploaded a new version (plus binaries) that already include the esp_init_data_default.

hanyuzzw, 2018-07-02:
Thanks for your reply,and if I want to test IPv6,where the lwip should I modify ,on `/esp-open-sdk/sdk`? on `/esp-open-sdk/esp-open-lwip`? on standalone sdk `ESP8266_NONOS_SDK-2.1.0-18-g61248df` or all of them?

martin-ger, 2018-07-02:
Neither the open-sdk nor the NONOS_SDK provide any working IPv6 support. Already tried to work on this, but this is a real heroic job...

If you want to start with IPv6 on the ESP8266, the FreeRTOS is probably the best choice. Here is a project of my students as a starting point:
https://github.com/IPv6-ESP8266/IPv6-ESP8266

hanyuzzw, 2018-07-02:
Thank you very much and I will try it.

yunfan, 2019-02-27:
i got the worse issues too
mine is a board Nodemcu lolin
and after i flashed the prebuilt binaries, i cant telnet to the target ip 192.168.240.1
also can not ping that ip

martin-ger, 2019-02-27:
I just checked the binary yesterday - for me it worked. Did you flash in DIO mode? Did you download the binaries with the browser? Sometimes they get corrupted by an CR/LF issue. Best is downloading the zip and extracting them.

The slip_router is a little bit hard to debug as it has no serial output. Just to check, whether flashing was all right, could you test to flash my "esp_mqtt" binariy  with the same procedure? Try to connect to the serial console there. If this fails too, you probably have an issue with flashing or downloading...

yunfan, 2019-02-28:
@martin-ger i guess i havnt use DIO mode, and i got the firmware by `git clone https`, might the problem you said

yunfan, 2019-02-28:
@martin-ger btw, the wifi in my local dont share the same ssid with the default firmware, is this caused the problem? and is it possible to dynmic configuring the ssid and password?

martin-ger, 2019-02-28:
`git clone` is fine. You can configure the SSID once you have access.

Did you do the following procedure?

To connect a Linux-based host, start the firmware on the ESP, connect it via serial to USB, and use the following commands on the host:
```
sudo slattach -p slip -s 115200 /dev/ttyUSB0&
sudo ifconfig sl0 192.168.240.2 pointopoint 192.168.240.1 up mtu 1500
```
now 
```
telnet 192.168.240.1 7777
```
gives you terminal access to the esp as router. On the ESP you then enter:

```
CMD>set ssid <your_ssid> 
CMD>set password <your_pw> 
CMD>set use_ap 0
CMD>save
CMD>reset
```
If so, when does it fail?

yunfan, 2019-02-28:
@martin-ger i had down slattatch and ifconfig 
the only problem is cant telnet to that address also cant ping that ip

martin-ger, 2019-02-28:
did you check, whether you can flash other firmwares correctly?

yunfan, 2019-03-01:
@martin-ger  yes, i'd flashed punyforth later, which works as expected

sardaukar, 2019-06-05:
I have the same problem :( I have one of these https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/overview and managed to flash the pre-built firmware on the repo, and got a connection via telnet once, but can't connect anymore. Please help.

sardaukar, 2019-06-05:
Oops figured it out - I changed the bitrate with `set bitrate` but was still doing `slattach` at 115200!

aladds, 2019-07-14:
So I am having the same issue as @yunfan.

I am using your pre-built binaries. I've tried with two different ESP boards (A NodeMCU "0.9" / ESP12 and an ESP-01)

I can make your esp wifi repeater code work correctly and connect to its wifi access point.

But this project...after flashing I get a constantly blinking LED on GPIO0 and ifconfig shows that whilst I am transmitting data, I never recieve any.

I have tried connecting with a USB cable to the NodeMCU, and I've tried a cp2102 bridge to both boards.

I made sure to download the zip file to avoid potential corruption issues. The SHA1 sums that I calculate match your files.

The esptool.py version I have is 1.3-dev (latest from pip) and I am using a Mac running 10.14.5. My slattach tests are being done on a Raspberry Pi

Any ideas of what else I can try?

aladds, 2019-07-14:
I have now tried again doing the whole process on the raspberry pi with esptool 2.6 with the same results.

GPIO0 flashing, no data transmitted.

aladds, 2019-07-14:
Default netmask!!! The default netmask is 255.255.255.255.

If you configure this to pretty much anything else it works.

martin-ger, 2019-07-14:
Where do you think it should be configured?

aladds, 2019-07-14:
ifconfig.

For some reason, the version of Raspbian I have (latest buster release on a Pi 4) gave me that default netmask for sl0, so I couldn't talk to anything.

I solved it with:

sudo ifconfig netmask 255.0.0.0

aladds, 2019-07-14:
Got it working with my PowerBook 540c, if you're interested in my use:

https://68kmla.org/forums/index.php?/topic/57516-esp8266-wireless-without-another-computer/

martin-ger, 2019-07-14:
This is great!

When I did this, I really wasn't aware of all the applications in retro computing. Now I have them all: Atari, Amiga, 286 PCs and a  PowerBook!