martin-ger/esp_slip_router #6: Slip Router starts then seems to reboot

lorneb, 2018-08-03:
Hi I'm using the latest bin files from the firmware folder dated 16 days ago (as of today)
on a esp8266 12F module, that has been running other firmware compiled from Arduino"for ages without issue. I have also tried loading the esp_init_data_default_v08_vdd33.bin @ 0x3FC000 but made no difference.  I'm using a raspberry pi2 with raspbian and configured it with the ones from your instructions relating to AP mode.

I get this on debug out put
SLIP Interface V1.0 starting

Config found and loaded

<<then it just repeats this over and over>>

if I watch the serial output @74880 I get then, over and over (delay of a few seconds between)
"@ªrjrA(¡‹SËu‹Pv›X¥(ý«Pyjuh)…"8át‡íÍ*NÈ~PÊTš1Æ¡e\ºËY@íY*M/=ání*E"HèIîA,R%¡ù¨E,R·*1)ÔvAm1Æ¡{~A,=ání*U"HhInA,tZ
år‡,R·*1)ÄvAí1Æ¡ûþA.=ání*e¢ÈèInA,ò%áÝîA,ò%árf cal sector: 1019
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.2.0-dev(9422289) compiled @ Nov  3 2017 19:40:08
phy ver: 1136_0, pp ver: 10.2

SOFTUART initialize gpio
SOFTUART bit_time is 52
SOFTUART TX INIT DONE
SOFTUART RX INIT DONE
SOFTUART INIT DONE

<<then it just repeats this over and over>>

Just after a few pointers to try thanks in advance

martin-ger, 2018-08-03:
Please do one HW (or power cycle) reset after flashing.

lorneb, 2018-08-03:
Ok I gave that a try, still the same seems to restart every 5 seconds.

this is the output from my Rpi.
root@raspberrypi:/var/log# slattach -d -v -L -p slip -s 115200 /dev/ttyUSB0
slattach: tty_open: looking for lock
slattach: tty_open: trying to open /dev/ttyUSB0
slattach: tty_open: /dev/ttyUSB0 (fd=3) slattach: tty_set_speed: 115200
slattach: tty_set_databits: 8
slattach: tty_set_stopbits: 1
slattach: tty_set_parity: N
slip started on /dev/ttyUSB0 interface sl0