martin-ger/esp_slip_router #7: NAT (napt) in AP mode

SergeyPL, 2019-02-25:
All works when  "use_ap 0", the 8266 is connected to other access point, slip channel connected to Linux PC. The linux has link to internet via 8266, port mapping is also work - forwarding the tcp port from STA ip to Linux PC.

But looks like the NAPT is not working when "use_ap 1".  Windows PC (192.168.4.2) connected to AP of 8266 (192.168.4.1), and starts connection to ssh of Linux PC (192.168.240.2). In linux PC tcpdump shows that source if incoming packets is Windows PC. 
I expect the NAT must replace the source ip (192.168.4.2) in packets, so Linux PC must  see the  source of 8622 (192.168.240.1).

I would be thanful for any explanation.

Next step i will try to compile all the sources and dubug (now i just upload the binary). 

PS. Mr. Martin-ger thank you for sharing this project.

martin-ger, 2019-02-25:
You are right, the NAPT in AP mode is not working. That's why I recommend to set the 
`sudo route add -net 192.168.4.0/24 gw 192.168.240.1`
This makes the Linux box answer to the requests.

SergeyPL, 2019-03-26:
if (config.use_ap) {
...
        // enable NAT on the AP interface
        IP4_ADDR(&my_ip, 192, 168, 4, 1);
        ip_napt_enable_no(0, 1);   // here the solution

    } ...

Port forwarding works. I connects from 192.168.4.2 (Windows station)  to 192.168.4.1 (esp AP) on port 22 (ssh) and esp forwards the packets to 192.168.240.2 (Linux station).

It allows hide the 192.168.240.* network from clients connected to espAP.

When it comes to "IP4_ADDR(&my_ip, 192, 168, 4, 1);", looks like the interface of esp do not have ip address assigned when ip_napt_enable(ip,1) is called.

closed 2019-03-26 by SergeyPL