martin-ger/esp_slip_router #9: from DOS to OTHER pc works - from OTHER to DOS not ping

PC-maniak, 2019-06-29:
Hi! i have problem .. from dos machine working ok .. ftp client to server connection ok, web browser ok
but from another machine in the network I have no chance in the connection .. I tried a real address ESP in my case (192.168.1.168) then also 192.168.240.2 ping, ftpsrv, not working at all .. what am I wrong?

martin-ger, 2019-06-30:
The internal addresses 192.168.240.x cannot work, as they are NATed and invisible to the outside world. However, you should be able to ping the external STA address of the ESP from other WiFi components.

To reach a service on the SLIP (192.168.240.2) machine you will need a portmap on the ESP, e.g. a web server:
`portmap add TCP 80 192.168.240.2 80`

Now you can reach it from the WiFi net with 'http://192.168.1.168'

PC-maniak, 2019-06-30:
Oh thank you !
it is a pity that the speed is so slow = (I would use it to FTP and transfer games to old laptops(386,486)
is there any way to use ESP32 and LPT (LIP) ?

martin-ger, 2019-06-30:
Actually, the main problem is not the power of the ESP but the speed of the serial connection. An ESP8266 can handle more than 5 mbps.

ssshake, 2021-12-17:
@PC-maniak can you tell me how you got DNS to work? I have this set up on a dos machine and I can ping and telnet to any host I want but I can't resolve any host names. I have tried multiple dns servers.