martin-ger/esp_slip_router #11: time exceeded in communication

Primy-games, 2019-09-17:
Hi, thanks for your work.
I am having trouble communicating the ESP8266 once I have introduced the precompiled firmware that it provides inside the firmware folder. Once the ESP8266 has been flashed from W7 correctly, I attach a log:

C:\Users\informatica.mam\Downloads\esptool-master\esptool-master>python esptool.
py --port COM5 write_flash -fs 8m 0x00000 firmware/0x00000.bin 0x10000 firmware/
0x10000.bin
WARNING: Flash size arguments in megabits like '8m' are deprecated.
Please use the equivalent size '1MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.8-dev
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
WARNING: Detected crystal freq 27.19MHz is quite different to normalized freq 26
MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: dc:4f:22:58:58:9c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0020
Compressed 38448 bytes to 26979...
Wrote 38448 bytes (26979 compressed) at 0x00000000 in 2.6 seconds (effective 117
.4 kbit/s)...
Hash of data verified.
Compressed 228644 bytes to 157900...
Wrote 228644 bytes (157900 compressed) at 0x00010000 in 17.5 seconds (effective
104.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

I cannot communicate with the module, I use telnet and Arduino IDE and it does not connect. I flash the ESP2866 again with the original firmware to verify that it works, and this time if it connects correctly. I try to flash with its firmware using different programs such as flash tool and receive the same behavior. Both python and esptool are well configured and the bariable PATH is also configured. what am I doing wrong? 
Greetings

martin-ger, 2019-09-17:
If it has flashed correctly, you have to connect to it using this procedure: https://github.com/martin-ger/esp_slip_router#usage-as-sta

Connection is via SLIP, not via serial console.

Primy-games, 2019-09-17:
Good evening, I'm afraid I can't communicate in any way possible, I attached the screenshot and the record so I can verify that there is no communication via SLIP.

screenshot:

https://1.bp.blogspot.com/-t9QmDoLsG6M/XYFLZZPZDvI/AAAAAAAABc4/Fmx_5f53-UYdMTg4yx0sUG2M_Cv_yiWTACLcBGAsYHQ/s1600/photo5766954877736563298.jpg

LOG:

esptool.py v2.6-beta1
Serial port COM3
esptool.py v2.6-beta1
Serial port COM3
esptool.py v2.6-beta1
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:18:6a:a3
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 128 bytes to 77...

Writing at 0x000fc000... (100 %)
Wrote 128 bytes (77 compressed) at 0x000fc000 in 0.0 seconds (effective 38.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
esptool.py v2.6-beta1
Serial port COM3
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:18:6a:a3
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0220
Compressed 38448 bytes to 26979...

Writing at 0x00000000... (50 %)
Writing at 0x00004000... (100 %)
Wrote 38448 bytes (26979 compressed) at 0x00000000 in 2.4 seconds (effective 127.4 kbit/s)...
Hash of data verified.
Compressed 228644 bytes to 157900...

Writing at 0x00010000... (10 %)
Writing at 0x00014000... (20 %)
Writing at 0x00018000... (30 %)
Writing at 0x0001c000... (40 %)
Writing at 0x00020000... (50 %)
Writing at 0x00024000... (60 %)
Writing at 0x00028000... (70 %)
Writing at 0x0002c000... (80 %)
Writing at 0x00030000... (90 %)
Writing at 0x00034000... (100 %)
Wrote 228644 bytes (157900 compressed) at 0x00010000 in 14.1 seconds (effective 129.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

martin-ger, 2019-09-18:
It seems that you are working on windows. Do you really have a SLIP setup on windows? If so, how does it lool like?

Primy-games, 2019-09-19:
I have also programmed ESP8266 with Ubuntu 14.04, 18.04, W7, Ubuntu 16.04 virtualized from W7 and the result is always the same I cannot communicate using its instructions:

sudo slattach -p slip -s 115200 /dev/ttyUSB0&
sudo ifconfig sl0 192.168.240.2 pointopoint 192.168.240.1 up mtu 1500
telnet 192.168.240.1 7777

A friend also tried using his firmware and gets the same result, he can't communicate.
I have repeatedly verified that everything was well configured, even from 4 different computers and two ESP8266, but the precompiled firmware does not allow me to communicate.

arananet, 2019-09-19:
Hi Martin, i'm Primy's friend. We are stuck on the same steps. We are unable to access via telnet to the device. All the step has been made as Primy comment on a linux (ubuntu) enviroment. We can see the usb - ttl device on linux under ttyUSBX (x is the number asigned by the machine).

So here is a resume:

ESP-01
EN pin to high (3v3).
IO0 to low (GND).

Flash the content of the prebuild firmware using the described command line on a Windows enviroment (others firmwares work OK and also others ESP too).

Remove the EN pin from the esp-01.

Run from a linux terminal:

sudo slattach -p slip -s 115200 /dev/ttyUSB0&
sudo ifconfig sl0 192.168.240.2 pointopoint 192.168.240.1 up mtu 1500
telnet 192.168.240.1 7777

Telnet connect and the USB-TTL is getting the keyboard strokes but, I dont have any CMD> line or any response.

I'have also tried using putty and setting up the telnet IP and PORT. No luck. With the arduino ide, listening on the usb-ttl port I get some info but is not related to the Slip firmware. 

Any tip?

martin-ger, 2019-09-19:
It might be the "puya-issue" - see also: https://github.com/martin-ger/esp_wifi_repeater/issues/336

Do you have any chance to test it with a different module, let's say a Wemos D1 mini?

Primy-games, 2019-09-23:
I bought a Wemos D1 mini as soon as I receive the esp8266 I tell him the experience, if that is the problem. Thank you

Primy-games, 2019-12-07:
Hi Martin-Ger.
After receiving the Wemos D1 mini, all the problems were solved, thank you very much for your help and your work.

illantalex, 2024-11-28:
I had the same issue flashing ESP-M3 (8285), it didn't have any telnet communication afterwards either. I managed to fix it by firstly erasing the flash:

`esptool.py --port /dev/ttyUSB0 --baud 115200 erase_flash`

And then programming it using the dio mode:

`esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash --flash_mode dio -fs 2M 0x00000 firmware/0x00000.bin 0x02000 firmware/0x02000.bin`