martin-ger/esp_slip_router #13: Two ESP8266 work with SLIP together as STA+AP

ElliottSeer, 2020-02-29:
Hi! Thank you for your work! It runs well on my  ESP8266. I am wondering that, is there any possible to drive two ESP8266 work with SLIP together as STA+AP, just like [github.com/martin-ger/esp_slip_router/issues/10](url). 
Because I am working on a porject about underwater communication, trying to link AP and STA with serial or something so people can use Wi-Fi provided by ESP8266 AP underwater without the distance limit. 
I was trying to link two ESP with SLIP, but I found that each ot them doing well but the STA didn't responded the data from AP.
[image]
So, could you give me some idea about how to fix this? THANKS!

martin-ger, 2020-02-29:
The first Problem is the fact, that it does NAT (even two times in this setup). So STAs cannot be reached actively from the internet in any case. However, the other direction should be possible, i.e. to allow STAs to connect to the internet (which is in most cases exactly what you want for a phone or pc).

The second problem is, that we would need NAT on the other interface for ESP2 ("use_as_ap 1", right?). I did some experiments with that, but finally skipped that (https://github.com/martin-ger/esp_slip_router/blob/ae8b9d234937400d3430a1e211ba96d080a7adce/user/user_main.c#L933 ) and decided to use a static route here. I don't know exactly, why I did it that way, but I had some issues here.

If you have a compiler environmet, you might try these lines. I might find some time to look into that, but don't expect me to look into that this weekend.

ElliottSeer, 2020-02-29:
LOL Thank you for your advice！ I do have the compiler environmet and I will try it !  And hope you  enjoy your weekend!

martin-ger, 2020-02-29:
If not already done, you definitly also have to change the IP-Address of one of the SLIP interfaces: default is 192.168.240.1 for both, that will obviously fail.

ElliottSeer, 2020-02-29:
> If not already done, you definitly also have to change the IP-Address of one of the SLIP interfaces: default is 192.168.240.1 for both, that will obviously fail.

Right I'm just trying to change the IP of one of the ESP (in config_flash.c 29 32 33, I changed 192.168.240.x to 192.168.233.x). But then I can't ping it. It seems that ESP did get the ping packets (the led blink) but my pc get no responses. But after I change it them back to 192.168.240.x, everything went on well. I didn't get it, did I miss some code which define the IP-Address?

ElliottSeer, 2020-02-29:
I tried to edit 
ip_addr = 192.168.240.2 
ip_addr_peer = 192.168.240.1
yet still didn't work
(Sorry to raise this issue at weekend, but I will take some more trying on it myself, don't worry)