martin-ger/esp_slip_router #15: Internet access via serial host's network interfaces

glanch, 2020-04-03:
I want to use esp_slip_router to isolate small footprint WiFi devices which's firmware I can't alter. The esp8266 will be attached to a Linux machine that has internet access via ethernet. Unfortunately these WiFi devices need internet access for a initialization. I want to give them controlled internet access via ethernet interface. Is it possible?
  `

martin-ger, 2020-04-04:
You use the ESP as AP? I think, you have to set up NAT routing on the Linux as well, so that it forwards packets coming from the SLIP interface to its external port.

glanch, 2020-04-04:
> You use the ESP as AP?
Yes I do.

> I think, you have to set up NAT routing on the Linux as well, so that it forwards packets coming from the SLIP interface to its external port.
For that the ESP needs to forward all packets addressed to Internet to the SLIP interface. Is this true?

junior013, 2022-09-19:
@martin-ger I know, that this issue is more than 2 years old, but I found your great code only now and I have the same problem.
It seems, that ESP in AP mode not forwarding packets to the SLIP peer if they are addressed to other subnet destination.
I use the precompiled firmware on the ESP and tcpdump the sl0 interface on linux side. If I ping 192.168.240.2 from a wifi client, i see the packets. But if I ping for example 8.8.8.8, I can't see any on the sl0 interface :(