martin-ger/esp_slip_router #18: STA mode - SSH connection refused

domas94, 2021-04-01:
ESP8266 is connected as STA to a wifi network. It got external address: 192.168.254.102.
When I try to SSH to that address I get response : Connection refused. 
When I try to telnet to that address with port 7777, connection is successful.
Any ideas? 
Thanks in advance.

martin-ger, 2021-04-01:
Think that is easy: the ESP has no SSH server.

domas94, 2021-04-01:
I am using ESP8266 as a WIFI interface for Raspberry3 compute module. So I actually want to connect to the Raspberry with SSH with a windows machine. Could you please provide an example of using a port forward?

martin-ger, 2021-04-01:
`portmap add tcp 22 192.169.240.2 22`
if you have the slip address of the raspi 192.169.240.2

domas94, 2021-04-07:
Thank you Martin. The SSH now works, but only when I disconnect the Ethernet connection from the RaspberryPi. Do you have any idea why is that?

martin-ger, 2021-04-07:
Should be a routing issue...

Look at the routing tab of the raspi before and after the ethernet disconnect. I guess, with ethernet the ssh-messages probably go to default route, which is not the SLIP interface. What if you define a static route back?

domas94, 2021-04-07:
[image]

This is my routing table. I tried to connect after deleting default eth0 route with:

ip route del default dev eth0

But didn't help. What do you mean by define a static route back?

martin-ger, 2021-04-07:
The ssh comes from which address?

domas94, 2021-04-07:
I am trying to ssh from my windows pc with ip address: 192.168.254.230 to esp8266 external ip address: 192.168.254.124.

martin-ger, 2021-04-07:
And the route for 192.168.254.0/23 goes via eth0.
Can you set a route 192.168.254.0 255.255.2550 to 192.168.240.1?

domas94, 2021-04-07:
It works when I type:

sudo ip route add 192.168.254.0/23 via 192.168.240.1

But then it blocks my ethernet connection.

martin-ger, 2021-04-07:
Try /24 - you have 192.168.254.0/24 in the WLAN and 192.168.255.0/24 in the ethernet, right?

domas94, 2021-04-07:
No, I have /23 also in ethernet. Netmask is 255.255.254.0

domas94, 2021-04-07:
Hmm, but I am not using this ip address 192.168.255.0/23. The problem is that when I add this route:

sudo ip route add 192.168.254.0/23 via 192.168.240.1
or
sudo ip route add 192.168.254.0/24 via 192.168.240.1

It disconnects my ethernet ssh connection. After I delete this route it works again.

martin-ger, 2021-04-07:
Perhaps I still don't get the whole setup: you have three network segments: the WiFi, the SLIP, and the ethernet. What are the network addresses of these networks?

domas94, 2021-04-07:
[image]

I have only 2 network segemnts, one is ethernet and the other is SLIP (esp8266). 
Ethernet network address: 192.168.254.220
Slip address: 192.168.240.2 
Slip gateway: 192.168.240.1
Slip external ip address: 192.168.254.124 (STA mode)

martin-ger, 2021-04-07:
But the STA is connected to a WiFi-Network with the Windows PC, right? This is the third network and you cannot have the sames address range in the WiFi and the Ethernet.

domas94, 2021-04-07:
Yes STA is connected to the same WIFI network as the Windows PC. Ok, thank you a lot for your time. I will try to figure something out.

closed 2021-04-07 by domas94

sreeshma-am, 2021-09-13:
Hi,

I am trying the same, to ssh my linux board which has slip interface via ESP32 module. The slip interface has IP 192.168.240.2 and station got 192.168.0.10 after connecting to an access point. If I try ping from a laptop in this same wifi network,  192.168.0.10 is responding. But ssh show connection refused error. I did portmap and also disconnected the default ssh interface the board already had. 

Please help me.