martin-ger/esp_slip_router #20: SSH to esp when there are multiple connections

domas94, 2021-04-09:
Hi Martin,

I will try to describe my problem once more, because I tried everything I could and couldn't get the solution. 

I used a classic raspberryPi3 and connected it to the same network with ethernet and WIFI. With that setup I could successfully SSH simultaneously to ethernet IP and WIFI IP address.

When I try to do the same with ethernet and slip ESP I get an error. SSH will only work on ethernet IP until I disconnect ethernet cable from the raspberry. It seems to me that the problem is because SLIP IP 192.168.240.2 doesn't belong to the network id where ethernet is connected. Because when I do

sudo ip route add 192.168.254.0/24 via 192.168.240.1

The SSH starts to work for the slip IP, but breaks for the ethernet IP. If I set the metric of sl0 higher than the metric of eth0, eth0 will start to work again, and vice versa.

Thank you for your time.

martin-ger, 2021-04-10:
It simply doesn't make sense to have two interfaces with different IP addresses from the same sub-network in different layer 2 networks. If you want to build a bridge, you will need a bridge device with one IP addess.

What do you want to achieve?

domas94, 2021-04-10:
I am developing software for an IoT gateway with RaspberryPi3 Compute module. It is important for us to SSH through WiFi since gateways are usually placed somewhere where they are hard to reach. I also tried SSH to WiFi when it is on a different sub-network that the ethernet. And again I had the same problem. WiFi was connected through mobile hotspot network, and ethernet on my private home network. SSH through WiFi started to work only when I disconnected ethernet cable.

martin-ger, 2021-04-10:
Could you send me a network plan with IP addess-ranges?

domas94, 2021-04-10:
[image]
[image]
[image]

I managed to ssh to raspberry via windows pc when I set the metric of sl0 lower than eth0. Only when sl0 and eth0 are on different networks.

domas94, 2021-04-10:
[image]
This is my network map

martin-ger, 2021-04-10:
When you set the default route on the raspi to 192.168.240.1 it should work, doesn't it?

You can also try just to set the route to the 192.168.126.0/24 net via  192.168.240.1 and leave 192.168.1.1 the default for anything else.

domas94, 2021-04-10:
If I add 192.168.126.0/24 net via 192.168.240.1 it works even if the sl0 metric is higher than eth0. I guess this should be ok for our purposes. Thank you once again for your help.

closed 2021-04-10 by domas94