martin-ger/esp_slip_router #24: Unable to build with custom build liblwip_open_napt.a

e1z0, 2022-03-03:
Hi, first of all i wanted to thank you for your software implementation it's really helpful and very useful in the vintage computers world!

But i've unable to build esp_slip_router with custom build esp-open-lwip. The lwip is built as described
```
Only if you don't want to use the precompiled library, checkout the sources from 
https://github.com/martin-ger/esp-open-lwip . Use it to replace the directory 
"esp-open-lwip" in the esp-open-sdk tree. "make clean" in the esp_open_lwip dir 
and once again a "make" in the upper esp_open_sdk directory. This will compile 
a liblwip_open.a that contains the NAT-features. 
Replace liblwip_open_napt.a with that binary.
```
I did that successfully, with standard **liblwip_open_napt.a** it builds ok, but if i try to build with custom build of "esp-open-lwip/liblwip_open.a" replaced as "liblwip_open_napt.a", i've got these errors:

```
root@9c35154a9cb0:~/esp_slip_router# make clean;make
CC driver/sio.c
CC driver/softuart.c
CC driver/uart.c
CC user/config_flash.c
CC user/ringbuf.c
CC user/user_main.c
AR build/app_app.a
LD build/app.out
build/app_app.a(user_main.o):(.text+0x8): undefined reference to `slipif_received_byte'
build/app_app.a(user_main.o): In function `write_to_pbuf':
/home/builder/esp_slip_router/user/user_main.c:827: undefined reference to `slipif_received_byte'
build/app_app.a(user_main.o): In function `console_handle_command':
/home/builder/esp_slip_router/user/user_main.c:422: undefined reference to `slipif_process_rxqueue'
/home/builder/esp_slip_router/user/user_main.c:445: undefined reference to `slipif_process_rxqueue'
/home/builder/esp_slip_router/user/user_main.c:518: undefined reference to `slipif_init'
collect2: error: ld returned 1 exit status
make: *** [build/app.out] Error 1
```

Thanks in advance!

martin-ger, 2022-03-03:
Make sure you use the correct Makefile: https://github.com/martin-ger/esp-open-lwip/blob/enc_polling/Makefile.open

It has to include: slipif.o ( see https://github.com/martin-ger/esp-open-lwip/blob/enc_polling/Makefile.open#L46 )

e1z0, 2022-03-04:
Yeah, i already tried to use Makefile.open and built library, then copied it to esp_slip_router folder and replaced the old one, but after trying to recompile esp_slip_router i've got that issue. I just wanted to implement fix in the library mentioned here: https://github.com/martin-ger/esp_slip_router/issues/17#issuecomment-1006925337

Thanks!

rohoog, 2024-09-28:
I had the same issues and only saw this issue after figuring out and fixing it myself, editing lwipopts.h and Makefile.open. Maybe the branch enc_polling of the repo https://github.com/martin-ger/esp-open-lwip.git should get these changes.
Note that copying the esp-open-lwip into the esp-open-sdk like the README says isn't necessary, you can just build the lwip library out-of-tree without problems.