martin-ger/esp_slip_router #25: Unable to telnet the ESP8266-01

palainp, 2022-03-09:
Not sure what's (not) going on, I'm currently unable to connect to the firmware with telnet (I do not change the wiring between the flash and the slattach):
```
$ esptool.py --port /dev/ttyUSB0 write_flash 0x00000 firmware/0x00000.bin 0x10000 firmware/0x10000.bin
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e8:db:84:df:2f:7f
Stub is already running. No upload is necessary.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00049fff...
Compressed 36848 bytes to 26072...
Wrote 36848 bytes (26072 compressed) at 0x00000000 in 3.0 seconds (effective 98.6 kbit/s)...
Hash of data verified.
Compressed 235760 bytes to 163144...
Wrote 235760 bytes (163144 compressed) at 0x00010000 in 14.6 seconds (effective 129.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
$ sudo slattach -L -p slip -s 115200 /dev/ttyUSB0&
[1] 1487
$ sudo ifconfig sl0 192.168.240.2 pointopoint 192.168.240.1 up mtu 1500
$ ip addr sh sl0
4: sl0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 10
    link/slip 
    inet 192.168.240.2 peer 192.168.240.1/32 scope global sl0
       valid_lft forever preferred_lft forever
$ ip route sh 192.168.240.1
192.168.240.1 dev sl0 proto kernel scope link src 192.168.240.2 
$ telnet 192.168.240.1 7777
Trying 192.168.240.1...
^C
```
I can see with wireshark that packets are going to 192.168.240.1 but I get no response. When the ESP is reset, I can see a quick flash from IO2 but no blue light afterwards. This is the same with the pre-compiled .bin files.

Any ideas ?

palainp, 2022-03-10:
Not sure if that is related but wth the following arguments it seems that the carrier signal is lost directly (I also tried to comment out the 2 GPIO_OUTPUT_SET (STATUS_LED, 1);` to get a blue led all the time but it failed, all is like the esp8266 simply does not start at all):
```
# time slattach -v -d -L -h -p slip -s 115200 /dev/ttyUSB0
slattach: tty_open: looking for lock
slattach: tty_open: trying to open /dev/ttyUSB0
slattach: tty_open: /dev/ttyUSB0 (fd=3)
slattach: tty_set_speed: 115200
slattach: tty_set_databits: 8
slattach: tty_set_stopbits: 1
slattach: tty_set_parity: N
slip started on /dev/ttyUSB0 interface sl0
slattach: tty_set_speed: 0

real	0m1,058s
user	0m0,000s
sys	0m0,003s
```

closed 2022-03-14 by palainp

spudje, 2025-08-30:
I'm running into the same problems. Can't get telnet to connect. Anybody solved this in the end?