martin-ger/esp_slip_router #27: Creating portmaps for port ranges (or 1-to-1 NAT, etc)

irrationalforest, 2022-07-20:
Hi there!

This is a fantastic project, thank you for your hard work.
It has helped get vintage many computers that don't have a NIC online.

I am wondering if there is any (undocumented?) way to add a portmap for a range of ports, or even all ports (complete 1-to-1 NAT style) rather than one at a time?

This is useful in situations such as running an FTP server on a machine, where the passive port range is typically at least 256 ports in size.
I have tried inputting a large number of portmaps via Expect in Linux, while connected via telnet, but it seems to hit a maximum limit.
Various other commands have also been tried, somewhat blindly:
portmap add TCP 0 192.168.240.2 0
portmap add TCP all 192.168.240.2 all
portmap add TCP any 192.168.240.2 any

These are all accepted, but do not function. It was perhaps a little silly to try. :)

martin-ger, 2022-07-21:
There is no implemented way to do this, but ist using FTP active not an option?

irrationalforest, 2022-08-19:
Hi Martin,

Thank you for the reply. Yes, using Active definitely does work.
I was hoping to use Passive since it allows others to access the FTP server without the need for re-configuration and a dedicated FTP client. Eg, using Windows Explorer.
That's OK, I'll find a way to get them using FileZilla in Active mode.
Thanks. :)

Kind regards,
Dominic Ashfield

closed 2022-08-19 by irrationalforest