martin-ger/esp_slip_router #28: esp_slip_router from windows 10

thecaver99, 2023-08-23:
i have flashed with windows 10
how can i set up the ip etc from windows as i dont have a linux box.

martin-ger, 2023-08-23:
Difficult as Windows has no SLIP-Driver - Linux VM?

thecaver99, 2023-08-23:
i did try with a usb boot drive but couldnt even work out how to get the make to work, so not even sure if i have the correct firmware flashed, could binaries not be made and released ready made, , sorry to sound like a real nooob at this but i am lol

thecaver99, 2023-08-23:
or is there a really noob guide on how to do this i.e. spoon fed

martin-ger, 2023-08-23:
What do you want to do with the slip_router? Only with a Windows PC it is probably useless for you...

martin-ger, 2023-08-23:
Hmm, you could enter the values as defaults in the code and rebuild the project, but this also requires Linux and is definitely more complex. I use VMware Workstation and an Ubuntu Image in my Windows 10...

thecaver99, 2023-08-24:
i tried again last night and managed to start the build but all i get was errors and cant get it to build.
is there any way to get the build made available to download pre-built instead.